team401/2025-Robot-Code #206: Tune climb better

aidnem, 2025-03-24:
**Purpose**
Add large feedforward to the climber to make it deploy faster. We should probably keep the big feedforward on the way down but we still need to deploy faster than we are deploying currently 

**Project Scope**
- [ ] Add constants for feedforward to java and json
- [ ] Make tuning mode for tuning them
- [ ] Tune them if we have time at shop

j0shuaS, 2025-03-26:
ok

j0shuaS, 2025-03-26:
@jkleiber it looks like FF variables already exist as well as a setFF method. i am a little confused
```java
public default void setFF(double kS, double kV, double kA, double kG) {}

  public final Double climbkS = 0.0;
  public final Double climbkV = 0.0;
  public final Double climbkA = 0.0;
  public final Double climbkG = 0.0;
//part of climb test mode
    LoggedTunableNumber.ifChanged(
            hashCode(),
            (ff) -> {
              io.setFF(ff[0], ff[1], ff[2], ff[3]);
            },
            climbkS,
            climbkV,
            climbkA,
            climbkG);
```
do we just need to tune them?

aidnem, 2025-03-28:
@j0shuaS new plan for this: Remove the logic where it conditionally applies the "feedforward" variable from the applyOutputs in ClimbIOTalonFX. Then, add that logic to climbState. Do something similar for searchingState except it only applies FF on the way up when it's more than a certain distance away from its setpoint.

jkleiber, 2025-03-28:
@aidnem I agree. From what I've heard, linear drive is an improvement on OTF and can be even better with proper tuning. However fine tuning should probably wait until the elevator is on

closed 2025-03-29 by avidraccoon