team401/2025-Robot-Code #207: Download better elevator belts

jkleiber, 2025-03-25:
# Summary
According to this meme account on CD, all we need to do is do more practice and  better software development to win DCMP: https://www.chiefdelphi.com/t/chs-batter-district-championship/496716

After reading this I realized that we forgot to write the software that makes the elevator belts unbreakable. We should add that patch in prior to DCMP 

As a drive by change we should also add the code to make sure the beelink never loses power and that it's Ethernet cable never unseats or gets cut. I know we've been holding back on this feature for awhile out of fear of becoming too goated and obtaining a 5 piece, but now it's time to really show the district what's what.

aidnem, 2025-03-25:
@jkleiber would this work

```java
// ElevatorConstants.java

public final Double beltStrength = 100000000.0
```

avidraccoon, 2025-03-25:
> @jkleiber would this work
> 
> ```java
> // ElevatorConstants.java
> 
> public final Double beltStrength = 100000000.0
> ```

Not sure if that is strong enough

jkleiber, 2025-03-25:
> @jkleiber would this work
> 
> ```java
> // ElevatorConstants.java
> 
> public final Double beltStrength = 100000000.0
> ```

We might as well go up to the maximum value of double

jkleiber, 2025-03-31:
new belts have been downloaded and installed on the robot, closing this very software related task

closed 2025-03-31 by jkleiber

aidnem, 2025-03-31:
@jkleiber closing without verification evidence, smh