team401/2025-Robot-Code #208: Investigate putting 2 beelinks on the robot

jkleiber, 2025-03-25:
Unironic ticket this time 

# Summary 

Our vision system has 2 key limitations:
1. Frame rate limit
2. Lack of redundancy

Both can theoretically be solved by putting 2 beelinks on the robot. 1:1 camera: coprocessor ratio would get us more FPS, and if one beelink fails to connect (cable cut, solder skill issue, etc.) we still have 1 camera

# work required 
- temporarily connect the backup beelink to comp bot via radio aux and set it up so both beelinks can communicate with the robot on different hostnames to avoid conflicts
- update vision code as needed to accept inputs from more than one coprocessor
- Demonstrate both co processors working to create a vision solution and figure out how much FPS we are getting with this config

If the FPS gain is notable we should switch to this configuration for DCMP. If the gains are minimal we may still keep the single coprocessor configuration because then we have a spare beelink

This is a fairly high priority and something the team can work while the elevator is being worked on in #207

aidnem, 2025-03-25:
@avidraccoon have you found anything online about interaction between 2 beelinks? I've been googling all day and haven't found much (except [this unhelpful CD post](https://www.chiefdelphi.com/t/photonvision-with-multiple-coprocessors/458364)). Everything else I've seen online has seemed to indicate that this is trivial, as I've seen it mentioned without any extra configuration required.

linglejack06, 2025-03-26:
@aidnem based off of the vision code i wrote and its docs, i dont think an additional coprocessor changes anything. I think we just need to make sure the configured name is correct

aidnem, 2025-03-26:
It has been investigated and no benefits were found :(

closed 2025-03-26 by aidnem