team401/2025-Robot-Code #210: Tune reef lineup (again)

jkleiber, 2025-03-25:
# Summary

After we do #208 and decide what to do there, we should tune the lineup controller again using teleop + smart autonomy to see if we can make it faster without inducing oscillations

Lineup time is our biggest time waste in auto and teleop right now, so minimizing it is our number 1 goal for DCMP imo

aidnem, 2025-03-25:
@jkleiber can we save this until after #201 is merged? I think whatever tuning we do would be obsolete after we have better control of the end of driving to lineup

jkleiber, 2025-03-25:
@aidnem yeah I view this as the capstone of that plus the multi-beelink setup

jkleiber, 2025-04-01:
Lineup looked fine yesterday once the cameras were the correct levels, so I'm going to deprioritize this and move it off DCMP milestone. If we qualify for worlds this may come back on the table so I'll leave the ticket open.