team401/2025-Robot-Code #225: Fix phase 1 to 2 overshoot in Linear Drive

aidnem, 2025-04-01:
**Purpose**
Currently linear drive ignores the fact that it has to turn between phases 1 and 2, so it blasts through the change at full speed ans completely misses. It needs to slow down before the end of phase 1.

**Project Scope**
- [ ] Figure out how to do this (maybe look at old mentor built branch)

aidnem, 2025-04-01:
I've fixed this on the mentor run branch; I'm going to branch off of that, interactive rebase off main, and then hopefully we can merge in this fix

Here's verification:

[image]

Yellow is setpoint, blue is actual; you can see that yellow is always under blue so we are always moving in the correct direction.

Here's drive velocity scalar, you can see it is always negative:

[image]

jkleiber, 2025-04-01:
@aidnem thanks for alleviating my concerns - this looks smooth. We should test this once we get the elevator setpoints locked in (I think we are probably okay on L2 and L3 for now, just need to get L4 and we can get the algae targets if needed when we do those PRs)

jkleiber, 2025-04-01:
The L4 setpoint we had last night was working decently with the coral fully pointing away from us. I think we may need to lift the elevator a bit higher but it should work well

aidnem, 2025-04-01:
Nice

aidnem, 2025-04-02:
One positive is that it looks like the overshoot was actually fixed by this branch

aidnem, 2025-04-02:
@jkleiber thanks for letting me know. How are autos/our ability to score looking?

jkleiber, 2025-04-02:
We can score and at the end of the night we probably would have had a 3 piece auto if the intake station was wider

closed 2025-04-02 by aidnem