team401/2025-Robot-Code #227: Pick an algae off the reef in auto

jkleiber, 2025-04-01:
# Summary

If we end up in a situation where we are on a goated allliance and are doing 1 piece barge, we should score the coral and then pick the algae off the reef and score it in our net

Doing this will accomplish a couple things:
- Net OTF (alliance dependent)
- Making it so the reef positions in auto actually do something

I think we'll need to upgrade our strategy manager schema to support this. Here's an idea:
```
"actions": [
    {
        "type": "AutoScore",
        "location": "Reef0",
        "level": "L4"
    },
    {
        "type": "AlgaeIntake",
        "location": "Algae0",
        "level": "AlgaeL2"
    },
    {
        "type": "AutoScore",
        "location": "Net",
        "level": "L4"
    }
]
``` 
This would allow us to be explicit about what the order of actions is and where they happen in a more general way than what we currently have. However it's probbaly a big refactor so we will need to test it extensively

@linglejack06 and @minhnguyenbhs assigning to y'all for now - feel free to split the work if it makes sense. I don't anticipate that we'll have this for DCMP, but I wouldn't be upset if we did lol

jkleiber, 2025-04-03:
@aidnem Preston wants this as an option at DCMP, which is a valid request. Maybe something you and @avidraccoon can work on on the bus ride?

aidnem, 2025-04-03:
@jkleiber Sure. I've already planned how to do it, I think to avoid major logic changes I'm going to add a flag that just replaces the auto with a preprogrammed sequence to do this. I don't trust rewriting all of our auto sequencing code on the day of load in.

jkleiber, 2025-04-03:
@aidnem fair, and sounds good

aidnem, 2025-04-03:
@linglejack06 any idea why drive desired location is getting set to reef 0 constantly? I added a new desired location for the auto line and I can see I'm setting it correctly but it gets immediately changed to reef0

linglejack06, 2025-04-03:
Probably snakecmscreen override check the set location thing in drive

aidnem, 2025-04-03:
It was, I fixed it

minhnguyenbhs, 2025-04-16:
[image]

lineup is blocking full auto testing in sim and i don't really know how to get around this. algae intake does not work yet, I'm working on it

minhnguyenbhs, 2025-04-16:
I think the issue might be it can't intake without scoring the existing piece? anyway I can't test the whole thing currently because it keeps getting stopped by lineup, but I've definitely seen it move correctly to everything except intake algae which I haven't been able to verify yet

aidnem, 2025-04-18:
@minhnguyenbhs I have working algae intake on my branch with #232 (and all of the rest of the commands required for this auto sequence). This includes a working (in sim) auto, it's just hard-coded. If we merged that branch could you implement my hardcoded auto with your strategy manager improvements?

closed 2025-04-23 by aidnem