team401/2025-Robot-Code #228: Use LEDs to indicate left or right pole in teleop

jkleiber, 2025-04-01:
## Summary

In teleop, let's turn all the LEDs one color if we are going to a right reef pole and a completely different color if we are going to a left pole. Then the driver will be able to see if a sidestep is needed much earlier

This only applies in teleop.

This is something I want to be able to test on Wednesday if it seems useful, so please have this ready by then. The only LED signal we currently understand on the drive team is purple = no beelink I think. So all other codes besides that one can be cleared out in favor of this request.

j0shuaS, 2025-04-01:
leds cooked D: so no leds

jkleiber, 2025-04-02:
Eventually LEDs may not be cooked. Reopening for that occasion

avidraccoon, 2025-04-02:
What's wrong with the leds?

j0shuaS, 2025-04-14:
close? idk if we are getting leds. still cooked and offseason now

closed 2025-04-14 by j0shuaS