team401/2025-Robot-Code #237: Add baby/demo mode (reduced speed for driving)

linglejack06, 2025-04-16:
**Purpose**
The purpose of this addition is to let new members and drive tryouts drive slower

**Project Scope**
- [ ] add json constant to slow down max speed

jkleiber, 2025-04-30:
Keeping this ticket open to make it possible to configure at runtime (I think we should use snakescreen to do this)

Thinking the following modes:
- Pro (full speed)
- Amateur (0.25 - 0.5x speed)
- Child (<= 0.1x speed)

j0shuaS, 2025-05-03:
hey @jkleiber  i was curious if we should have a whole other config file for just changing the max linear and angular speeds or if we should do your approach(which i am working on right now) or if we should have some combo of the two where you can only change the speed in demo mode? i am not sure. I also was curious about the actual way of doing it. Aiden and I talked and we think we should have a constant file that holds all of that information(ei. max linear and angular speeds for each level) and then when you pick one it just subs that in. thxs :D

j0shuaS, 2025-05-04:
figured out how to do. big chilling now

blandreth94, 2025-05-05:
We should consider adding feature disable functions, ie disable elevator, hood, etc toggles.

Also add elevator height limit in every demo mode that does not go over 6' total height to prevent tent contact.